orbbec/OrbbecSDK_ROS2 #143: 使用orbbecsdk_ros2的gemini435le.launch.py出现问题

rokkieluo, 2025-09-20:
我在使用orbbecsdk_ros2的gemini435le.launch.py出现问题
当我在运行ros2 launch orbbec_camera gemini435le.launch.py时会出现打不开的相机的情况
终端报错如下：
cat@lubancat:~/orbbec_ros2_ws$ ros2 launch orbbec_camera gemini435_le.launch.py 
[INFO] [launch]: All log files can be found below /home/cat/.ros/log/2025-09-20-16-29-43-615495-lubancat-701678
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [701757]
[component_container-1] [INFO] [1758356984.469566040] [camera.camera_container]: Load Library: /home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/liborbbec_camera.so
[component_container-1] [INFO] [1758356984.651435261] [camera.camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1758356984.654557160] [camera.camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<orbbec_camera::OBCameraNodeDriver>
[component_container-1] [INFO] [1758356984.707551577] [camera.camera]: setUvcBackendType:libuvc
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/camera/camera_container'
[component_container-1] [INFO] [1758356985.721442642] [camera.camera]: Connecting to device with net ip: 192.168.1.10
[component_container-1] [INFO] [1758356985.721591096] [camera.camera]: Before lock: Select device net ip: 192.168.1.10
[component_container-1] [INFO] [1758356985.721615303] [camera.camera]: After lock: Select device net ip: 192.168.1.10
[component_container-1] [INFO] [1758356985.721657594] [camera.camera]: FindDeviceByNetIP device net ip 192.168.1.10
[component_container-1] [INFO] [1758356985.721684426] [camera.camera]: getDeviceByNetIP device net ip 192.168.1.10 done
[component_container-1] [INFO] [1758356988.692621485] [camera.camera]: Select device cost 2971 ms
[component_container-1] [INFO] [1758356988.692991306] [camera.camera]: Try to connect device via Ethernet
[component_container-1] [INFO] [1758356988.693619827] [camera.camera]: OBCameraNode: use_intra_process: OFF
[component_container-1] [INFO] [1758356988.737162257] [camera.camera]: current time domain: global
[component_container-1] [INFO] [1758356988.737354751] [camera.camera]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000
[component_container-1] 
[component_container-1] [INFO] [1758356988.737420374] [camera.camera]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000
[component_container-1] 
[component_container-1] [INFO] [1758356988.737469664] [camera.camera]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 17000
[component_container-1] 
[component_container-1] [INFO] [1758356988.737513996] [camera.camera]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 30000
[component_container-1] 
[component_container-1] [INFO] [1758356989.143032881] [camera.camera]: Setting heartbeat to OFF
[component_container-1] [INFO] [1758356989.143368287] [camera.camera]: Create align filter
[component_container-1] [INFO] [1758356989.144235675] [camera.camera]: Depth process is HW
[component_container-1] [INFO] [1758356989.144298382] [camera.camera]: Setting LDP to ON
[component_container-1] [INFO] [1758356989.150789214] [camera.camera]: Setting G300 laser control to 1
[component_container-1] [INFO] [1758356989.154032151] [camera.camera]: Setting laser control to 1
[component_container-1] [INFO] [1758356989.157045262] [camera.camera]: Available presets:
[component_container-1] [INFO] [1758356989.158086187] [camera.camera]: Load device preset: Default
[component_container-1] [ERROR] [1758356989.159997709] [camera.camera]: Error in setupDevices at line 311: Component 21 not found!
[component_container-1] [ERROR] [1758356989.164822012] [camera.camera]: Failed to load device preset: Component 21 not found!
[component_container-1] [INFO] [1758356989.169547151] [camera.camera]: Current sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1758356989.217095077] [camera.camera]: Set sync mode: OB_MULTI_DEVICE_SYNC_MODE_STANDALONE
[component_container-1] [INFO] [1758356989.217405692] [camera.camera]: Setting color mirror to OFF
[component_container-1] [INFO] [1758356989.217633768] [camera.camera]: Setting color flip to OFF
[component_container-1] [INFO] [1758356989.217871177] [camera.camera]: Setting depth mirror to OFF
[component_container-1] [INFO] [1758356989.217976756] [camera.camera]: Setting depth flip to OFF
[component_container-1] [INFO] [1758356989.218082628] [camera.camera]: Setting noise removal filter:ON
[component_container-1] [INFO] [1758356989.218141543] [camera.camera]: Setting color auto white balance to ON
[component_container-1] [INFO] [1758356989.221019617] [camera.camera]: Setting color auto exposure to ON
[component_container-1] [INFO] [1758356989.224434423] [camera.camera]: Setting depth auto exposure priority to OFF
[component_container-1] [INFO] [1758356989.228200302] [camera.camera]: Setting IR auto exposure to ON
[component_container-1] [INFO] [1758356989.563819440] [camera.camera]: default noise removal filter min diff: 512
[component_container-1] [INFO] [1758356989.564244093] [camera.camera]: after set noise removal filter min diff: 256
[component_container-1] [INFO] [1758356989.564350840] [camera.camera]: default noise removal filter max size: 40
[component_container-1] [INFO] [1758356989.564493460] [camera.camera]: after set noise removal filter max size: 80
[component_container-1] [INFO] [1758356989.564894197] [camera.camera]: Setting accel data correction to ON
[component_container-1] [INFO] [1758356989.564986361] [camera.camera]: Setting gyro data correction to ON
[component_container-1] [INFO] [1758356989.566300860] [camera.camera]: Setting DecimationFilter......
[component_container-1] [INFO] [1758356989.566414023] [camera.camera]: set DecimationFilter to false
[component_container-1] [INFO] [1758356989.566582018] [camera.camera]: Skip setting filter: DecimationFilter
[component_container-1] [INFO] [1758356989.566663974] [camera.camera]: Setting HDRMerge......
[component_container-1] [INFO] [1758356989.566722888] [camera.camera]: set HDRMerge to false
[component_container-1] [INFO] [1758356989.566781803] [camera.camera]: Skip setting filter: HDRMerge
[component_container-1] [INFO] [1758356989.566849176] [camera.camera]: Setting SequenceIdFilter......
[component_container-1] [INFO] [1758356989.566906924] [camera.camera]: set SequenceIdFilter to false
[component_container-1] [INFO] [1758356989.566965839] [camera.camera]: Skip setting filter: SequenceIdFilter
[component_container-1] [INFO] [1758356989.567034962] [camera.camera]: Setting SpatialAdvancedFilter......
[component_container-1] [INFO] [1758356989.567091543] [camera.camera]: set SpatialAdvancedFilter to false
[component_container-1] [INFO] [1758356989.567151041] [camera.camera]: Skip setting filter: SpatialAdvancedFilter
[component_container-1] [INFO] [1758356989.567218706] [camera.camera]: Setting TemporalFilter......
[component_container-1] [INFO] [1758356989.567272246] [camera.camera]: set TemporalFilter to false
[... 376 lines cut ...]
#3    Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/libOrbbecSDK.so.2", at 0x7fb031ff27, in 
#2    Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/libOrbbecSDK.so.2", at 0x7fb0338704, in 
#1    Object "linux-vdso.so.1", at 0x7fb75f87af, in 
#0    Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/liborbbec_camera.so", at 0x7fb0dccc6f, in signalHandler(int)
Stack trace (most recent call last) in thread 701933:
#19   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#18   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb7015edb, in 
#17   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb6fad5b7, in 
#16   Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7fb71e29db, in 
#15   Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/libOrbbecSDK.so.2", at 0x7fb033e69f, in 
#14   Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/libOrbbecSDK.so.2", at 0x7fb025bc73, in 
[... cut ...]
#5    Object "/lib/aarch64-linux-gnu/libstdc++.so.6", at 0x7fb71b53cf, in __gnu_cxx::__verbose_terminate_handler()
#4    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb6f5712f, in abort
#3    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb6f6a67b, in raise
#2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb6faf1f0, in 
#1    Object "linux-vdso.so.1", at 0x7fb75f87af, in 
#0    Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/liborbbec_camera.so", at 0x7fb0dccc6f, in signalHandler(int)

winf, 2025-09-26:
launch里面的color_fps不能设成20，设备在640*400分辨率下不支持这个帧率，请改成支持的帧率，depth_fps同理。如何确定相机支持的帧率可以用下面这个指令确定，`ros2 run  orbbec_camera list_camera_profile_mode_node`

rokkieluo, 2025-09-26:
> launch里面的color_fps不能设成20，设备在640*400分辨率下不支持这个帧率，请改成支持的帧率，depth_fps同理。如何确定相机支持的帧率可以用下面这个指令确定，`ros2 run orbbec_camera list_camera_profile_mode_node`

当我使用ros2 run  orbbec_camera list_camera_profile_mode_node后，终端出现报错：
cat@lubancat:~/orbbec_ros2_ws$ ros2 run  orbbec_camera list_camera_profile_mode_node
terminate called after throwing an instance of 'ob::Error'
  what():  No device found
Stack trace (most recent call last):
#12   Object "/usr/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1", at 0xffffffffffffffff, in 
#11   Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/orbbec_camera/list_camera_profile_mode_node", at 0x555bf0a42f, in _start
#10   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7fb12a74cb]
#9    Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __aarch64_ldadd4_acq [0x7fb12a73fb]
#8    Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/orbbec_camera/list_camera_profile_mode_node", at 0x555bf09baf, in main
#7    Object "/home/cat/orbbec_ros2_ws/install/orbbec_camera/lib/orbbec_camera/list_camera_profile_mode_node", at 0x555bf0993f, in ob::Error::handle(ob_error**, bool) [clone .part.0]
#6    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb1502e93, in __cxa_throw
#5    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb1502baf, in std::terminate()
#4    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb1502b4b, in 
#3    Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb15053cf, in __gnu_cxx::__verbose_terminate_handler()
#2    Source "./stdlib/abort.c", line 79, in abort [0x7fb12a712f]
#1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7fb12ba67b]
#0    Source "./nptl/pthread_kill.c", line 44, in __pthread_kill_implementation [0x7fb12ff1f0]
Aborted (Signal sent by tkill() 51963 1000)
[ros2run]: Aborted